FOME-Tech/fome-fw #93: TPS MAP fallback table has no axis as default

OrchardPerformance, 2023-05-02:
Thank you for taking the time to report an issue!

To make our work easier, please provide as many details as you can:
* [ ] What hardware did you use while experiencing the issue?

mck1117, 2023-05-02:
Can you provide some more context? Not sure what this is asking.

OrchardPerformance, 2023-05-04:
[image]
This is how it is on a fresh firmware flash at the moment. 

No values for either X or Y axis.

mck1117, 2023-05-04:
Much better context, thank you!

nmschulte, 2023-05-04:
Mmm, not sure what's going on here; this is what Proteus F7 build shows on new project load:
[image]

-21474836.48

mck1117, 2023-05-04:
Hopefully eventually we delete that table and use #63 instead :)

closed 2023-05-04 by mck1117

OrchardPerformance, 2023-05-04:
Is it bad I'm ok with the idea of killing that table down to a 4x4? With a
view to obsolete it quickly?

PR #96 (2023-05-04): set default MAP estimate axis values
fixes #93