FOME-Tech/fome-fw #102: Double solenoid ICV mode - Open loop open % is doubled

OrchardPerformance, 2023-05-06:
Using the slider for % opening of the IAC causes the commanded % to be doubled. 

10% causes 20% duty cycle on the open solenoid, 40% causes 80% duty, any position greater than 50% saturates the output. 
The close duty responds inverse so it's working as expected just double. 

Checked on our current release, verified on the MCU pins with a scope. 

Low priority for now, Will confirm function is ok on BMW and advise any users duty is doubled in current release.

mck1117, 2023-05-07:
What do we expect double solenoid mode to do? Can you fill in this table?

|Logical Idle %|Solenoid A %|Solenoid B %|
|----|----|----|
|0| | |
|25| | |
|50| | |
|75| | |
|100| | |

classicdaily, 2023-05-07:
Is this actual output or output you see in TS?  If actual output have you tried changing HZ?  I m running RE and double solenoid mode works fine for me.  It did take some testing with different HZ levels.

OrchardPerformance, 2023-05-07:
This is oscilloscope measured duty at the trigger pin for the lowsides.
300hz, same a I was using with one side pulled up and the other pulsed.

I will grab some data off the Z for cross checking.

classicdaily, 2023-05-07:
Gotcha.  I know on the BMW valve that I am using.  At 300hz there is VERY little resolution.  50-70% duty would make the valve open all the way.  Run it at 70hz and now duty matches actual opening of the valve spot on.

OrchardPerformance, 2023-05-07:
Right, looks like with proper investigation with a spare valve on car it's not an output scale issue, the output scale may have been a result of the tune I was using to test. 

I plugged in and scoped the actual outputs on car making sure all modifiers were off and 0-100% does give 0-100%. 

However Bosch solenoid does not operate over the full range. Regardless of frequency the solenoids are closed at ~25% and fully open at 75%. 

2 options: 
1. Wiki entry that valves will not function right below 25% and above 75%. 
2. Bosch mode option that caps the range to be 0 =25%  and 100 =75%

mck1117, 2023-05-07:
I've previously thought about an "output rescale" mode to fix stuff like that. Similarly, Volvo idle valves are backwards: more duty closes the valve.

mck1117, 2023-05-07:
> Gotcha. I know on the BMW valve that I am using. At 300hz there is VERY little resolution. 50-70% duty would make the valve open all the way. Run it at 70hz and now duty matches actual opening of the valve spot on.

Are you running it on channels with freewheel diodes? They're required for linear operation of solenoids like idle, VVT, etc.

OrchardPerformance, 2023-05-07:
@classicdaily  - I will try it again at 700hz but saw no change from 300-500hz. 

For what it's worth I was running 300hz before in single wire with the close side pulled down using a big resistor (old school MS style).

classicdaily, 2023-05-07:
I am running mine at 70hz NOT 700.