FOME-Tech/fome-fw #144: Broken ts tips after auto log field generation change

OrchardPerformance, 2023-06-19:


OrchardPerformance, 2023-07-04:
@nmschulte poke so we can close this if it's a non-issue.

nmschulte, 2023-07-04:
I'll get at it this week; my ECU's been in garage purgatory.

I believe this is still an issue, but I forget the details.

nmschulte, 2023-07-05:
> @nmschulte is this still true?

The issue I reported there is fixed via commits referenced in https://github.com/rusefi/rusefi/issues/4927.

Closing this now.

nmschulte, 2023-07-05:
ex:

> This is just my observation from e.g. 5 ft pole -- perhaps I misunderstand the intent behind these comments.

Some specific examples, though pretty much any config in rusefi_config.txt with a new-line seems like it'd be useful to have the full comment in TunerStudio:

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L302

```
float baseFuel;Base mass of the per-cylinder fuel injected during cranking. This is then modified by the multipliers for CLT, IAT, TPS ect, to give the final cranking pulse width.\nA reasonable starting point is 60mg per liter per cylinder.\nex: 2 liter 4 cyl = 500cc/cyl, so 30mg cranking fuel.;"mg", 1, 0, 0, 500, 1
int16_t rpm;This sets the RPM limit below which the ECU will use cranking fuel and ignition logic, typically this is around 350-450rpm. \nset cranking_rpm X;"RPM", 1, 0, 0, 3000, 0
end_struct
```

[image]

closed 2023-07-05 by nmschulte