FOME-Tech/fome-fw #176: Does VVT_FIRST_HALF need to exist?

mck1117, 2023-06-28:
VVT_FIRST_HALF and VVT_SECOND_HALF do the same thing, and you can get one from the other by adding 360 to your trigger offset.

The problem with VVT_FIRST_HALF is that it can have unpredictable results on non-360 deg primary triggers, like 12 tooth crank for example.

VVT_SECOND_HALF:

(TDC is last primary tooth before cam tooth)

[image]

VVT_FIRST_HALF:

(TDC is _two_ teeth before cam tooth, that's odd?)

[image]

Since these two modes do fundamentally the same thing, best option is maybe:

- VVT_FIRST_HALF -> delete
- VVT_SECOND_HALF -> VVT_SINGLE_TOOTH

mck1117, 2023-06-29:
tangentially related 7ae11160d9bfb31a664130e66586e849b8201c8a

OrchardPerformance, 2023-07-04:
Yep, I would tend to agree with that. I don't see why we need to treat the cam sync differently for 1st crank 360 vs 2nd crank 360. 
I have not seen it differentiated in anything else, normality is to assume 0 is TDC firing stroke and 720 is the max accepted offset. 

As long as we note to users in the wiki that some VVT offsets may need 360 degrees adding to the offset then I see no issue.

mck1117, 2024-02-20:
VVT first/second half is gone, it's just single tooth mode now.

closed 2024-02-20 by mck1117