FOME-Tech/fome-fw #180: F7 update without full erase

rusefillc, 2023-06-30:
ST prog tool has a bug https://community.st.com/t5/stm32cubeprogrammer-mcu/stm32cubeprogrammer-incorrectly-erases-on-dual-bank-1mb-stm32f7/m-p/127960

What kinds of workarounds are possible in order to use 768kb for firmware but not erase calibrations?

rusefillc, 2023-06-30:
What is the exact nature of ST bug anyway? We know it does not skip sector indices while erase operation but it seems to reference correct addresses, why does sector index even matter?

rusefillc, 2023-06-30:
Console already communicates with firmware to initiate DFU reboot. What if we have firmware erase 512k-768k sectors programmatically? Would that allow ST tool to properly update?

Well, only problem such self-erasing code would have to be located outside of the 512k-768k range and that starts to look like a custom bootloader already :(

nmschulte, 2023-06-30:
If I understand correctly, this issue isn't just stm32cubeprogrammer, it also affects `dfu-util`, which to me means it's something about the chip or the vendor firmware.

rusefillc, 2023-06-30:
I would assume that ``dfu-util`` simply does not handle the odd corner case correctly. 

See for instance https://github.com/feaser/openblt/blob/afd5cb10d71d00708d761fb796efe0e2e2ef776b/Target/Source/ARMCM7_STM32F7/flash.c#L139 yet another example of a piece of software without proper logic of dual bank layout

nmschulte, 2023-06-30:
Indeed it seems like `dfu-util` does not do _any_ layout concern: it just does what the DfuSe file says.  Perhaps we are simply not building the `.dfu` correctly for the given target/layout (possibly via some black box, that we might be missing a control to).

mck1117, 2023-06-30:
> What kinds of workarounds are possible in order to use 768kb for firmware but not erase calibrations?

custom bootloader

nmschulte, 2023-06-30:
> I would assume that `dfu-util` simply does not handle the odd corner case correctly.

I guess you're right: https://sourceforge.net/p/dfu-util/dfu-util/ci/master/tree/src/dfuse_mem.c#l82

mck1117, 2023-06-30:
> What if we have firmware erase 512k-768k sectors programmatically? Would that allow ST tool to properly update?

If you do this, the firmware will pretty much instantly hard fault as it tries to execute code (now all FFFF) above 512k.

mck1117, 2023-06-30:
> but not erase calibrations

My expectation is that it's likely for the config shape to change every public release (https://github.com/fome-tech/fome-fw/releases), so this is not really a big deal. The ECU is 5 seconds away from ignoring the saved calibration anyway and resetting to defaults.

mck1117, 2024-02-20:
This is fixed by updating with OpenBLT.

closed 2024-02-20 by mck1117