FOME-Tech/fome-fw #183: release package includes outdated version 2.12.0 of STM32_Programmer_CLI

mi-hol, 2023-07-03:
\fome.release_2305.mre_f4\console\ includes version 2.12 of STM32_Programmer_CLI

1. version is outdated and was replaced by 2.12.1
2. [conversations in discord ](https://discord.com/channels/1060875162892898324/1100956391608434719/1124464612113391626) indicate that dfu_util is the preferred flashing app

Question:
Should it be removed from or upgraded in release package?

nmschulte, 2023-07-03:
v2.13 (and v2.13.1 for MacOS) exist now too: https://www.st.com/en/development-tools/stm32cubeprog.html#st-get-software

I was about to put up a PR to update this, but it seems I cannot get at the goods as I'm not on Window (trouble installing all the goods under Wine).

I do wonder ... do we have license to redistribute this, and how do we disambiguate its license vs FOME's GPL?

https://www.st.com/resource/en/license/SLA0048_STM32CubeProg.pdf

> 2. **Redistributions in binary form**, except as embedded into microcontroller or microprocessor device manufactured by or for STMicroelectronics or a software update for such device, **must reproduce the above copyright notice, this list of conditions and the following disclaimer in the documentation and/or other materials provided with the distribution.**

> 5. No use, reproduction or redistribution of this software package partially or totally may be done in any manner that would subject this software package to any Open Source Terms (as defined below).

mi-hol, 2023-07-03:
@nmschulte thanks for digging into license terms.
With that the only options seems "remove f STM32_Programmer_CLI" from release package.

Any other views?

nmschulte, 2023-07-03:
I think we can distribute (even parts of) it, but we'd need to make clear the copyright notice and conditions and disclaimer etc.

closed 2023-08-03 by mck1117