FOME-Tech/fome-fw #192: Knock Control : Usability of max allowed retard table

OrchardPerformance, 2023-07-17:
Current max allowed knock retard table is just a table of degrees, requiring a user to subtract this from their ignition table to check if they are going to be past TDC/too retarded. 

We should make this a table of degrees relative to TDC, so a user only has to enter their desired worst case ignition timing. 

Current method will work, just requires more calculation hence the enhancement tag. 

[image]

mck1117, 2023-09-15:
Should this change if we still have a global min/max timing setting?

OrchardPerformance, 2023-09-25:
Nahh, it's a pretty application specific requirement, we are trying to prevent insane egts as well as firing past tdc on wasted spark so it's quite load/rpm dependent.

OrchardPerformance, 2023-11-05:
Played about with this the other day, needs a tool tip but that's a minor thing. 
Will create a tool tip ticket when we are sure it all works

closed 2023-11-05 by OrchardPerformance