FOME-Tech/fome-fw #193: Knock control : Reapply rate as % of aggression

OrchardPerformance, 2023-07-17:
Current method of degrees/second is too slow at low load and too fast at high load. (also hard to pick a reasonable base value).

The aggression was designed to vary the size of the retard steps with the ignition advance, using a % of aggression to do the reapply will also allow it to scale with the size of the retard steps. 

Suggest we use a base value of about 30% of the aggression value, that way we will re-advance at ~1/3rd any step calculated for retard. 

[image]

mck1117, 2023-09-15:
Thoughts on just using a small load/rpm table for reapply? That seems to be what the OEMs do.

OrchardPerformance, 2023-09-16:
Shouldn't actually be needed, using the % of aggression will mean the reapply rate is varied to match the ignition timing at that load/rpm. 
It is likely a default of 0.3 will work for most cases. 

What we may need is the inverse and for the retard aggression to be a small load/rpm table if we find it's hard to get just the right amount.