FOME-Tech/fome-fw #195: java console build reports "deprecated API" and "unchecked or unsafe operations"

mi-hol, 2023-07-17:
FOME console / build (pull_request)  
[image]

**Solution:**
change compile options to include:
```
-Xlint:unchecked 
-Xlint:unchecked 
```

I'd be happy to create a PR to implement these recommendations.
@mck1117 please assign to me

mck1117, 2023-07-17:
are the unsafe or unchecked operations needed or not needed? ie, is the compiler trying to tell us something we should listen to instead of silencing it?

mi-hol, 2023-07-17:
from https://stackoverflow.com/questions/12558231/recompile-with-xlintdeprecation-for-details to my understanding "the compiler will tell us more details we should fix" when these options are added

mck1117, 2023-07-17:
when you run a build adding those options, what happens?

mi-hol, 2023-07-17:
I'm before the stage of having the change made, assignment of issue was intended to confirm you are ok with me trying it out

mi-hol, 2023-07-17:
see below screenshot for an example
see https://github.com/mi-hol/fome-fw/actions/runs/5576099435/jobs/10186895169 for all results
`    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 1 error
    [javac] 99 warnings

BUILD FAILED
/home/runner/work/fome-fw/fome-fw/java_console/build.xml:58: Compile failed; see the compiler error output for details.

Total time: 4 seconds
Error: Process completed with exit code 1.`
[image]

mi-hol, 2023-07-17:
condensed errors for "unchecked" warning is in https://github.com/FOME-Tech/fome-fw/issues/196

mck1117, 2023-07-19:
options added in [cbcc1ef](https://github.com/FOME-Tech/fome-fw/commit/cbcc1ef254fd460955f1eda480a70158fccfa09d)

mi-hol, 2023-07-20:
doesn't fully work yet. For details see https://github.com/FOME-Tech/fome-fw/issues/197#issuecomment-1643473306

mi-hol, 2023-07-21:
@mck1117 please reopen while figuring out why it doesn't work on master branch

mi-hol, 2025-01-06:
works

closed 2025-01-06 by mi-hol