FOME-Tech/fome-fw #197: FOME console 92 instances of warning: [deprecation]

mi-hol, 2023-07-17:
```
Search "warning: [deprecation]" (92 hits in 1 file of 1 searched)
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractIntegerDistribution.java:27: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractRealDistribution.java:28: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/ui/src/main/java/com/rusefi/SerialPortScanner.java:12: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/widgets/AnyCommand.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/file/BaseMap.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/YAxisData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserBaseListener.java:7: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserListener.java:5: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/maintenance/VersionChecker.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/AverageData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYDataReader.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/MessagesView.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/WarningPanel.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/config/BaseConfigField.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/util/DefaultExceptionHandler.java:4: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractIntegerDistribution.java:27: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractRealDistribution.java:28: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/ui/src/main/java/com/rusefi/SerialPortScanner.java:12: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/widgets/AnyCommand.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/file/BaseMap.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/YAxisData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserBaseListener.java:7: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserListener.java:5: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/maintenance/VersionChecker.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/AverageData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYDataReader.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/MessagesView.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/WarningPanel.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/config/BaseConfigField.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/util/DefaultExceptionHandler.java:4: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractIntegerDistribution.java:27: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/AbstractRealDistribution.java:28: warning: [deprecation] RandomDataImpl in org.apache.commons.math3.random has been deprecated
/java_console/ui/src/main/java/com/rusefi/SerialPortScanner.java:12: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/widgets/AnyCommand.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/file/BaseMap.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/YAxisData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserBaseListener.java:7: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserListener.java:5: warning: [deprecation] NotNull in org.antlr.v4.runtime.misc has been deprecated
/java_console/ui/src/main/java/com/rusefi/maintenance/VersionChecker.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/AverageData.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/models/XYDataReader.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/MessagesView.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/WarningPanel.java:3: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/config/BaseConfigField.java:5: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/ui/src/main/java/com/rusefi/ui/util/DefaultExceptionHandler.java:4: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/IoUtil.java:115: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/IoUtil.java:153: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:22: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:37: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:44: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:45: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:46: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/EnduranceTestUtility.java:49: warning: [deprecation] FileLog in com.rusefi has been deprecated
/java_console/autotest/src/main/java/com/rusefi/SimulatorExecHelper.java:25: warning: [deprecation] FileLog in com.rusefi has been deprecated
[... cut ...]
/java_console/models/src/main/java/org/apache/commons/math3/distribution/TDistribution.java:253: warning: [deprecation] isSupportLowerBoundInclusive() in RealDistribution has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/WeibullDistribution.java:338: warning: [deprecation] isSupportUpperBoundInclusive() in RealDistribution has been deprecated
/java_console/models/src/main/java/org/apache/commons/math3/distribution/WeibullDistribution.java:333: warning: [deprecation] isSupportLowerBoundInclusive() in RealDistribution has been deprecated

```

mi-hol, 2023-07-17:
Affected are;

- ANTLRInputStream in org.antlr.v4.runtime
- FileLog in com.rusefi
- isSupportUpperBoundInclusive() in RealDistribution
- NotNull in org.antlr.v4.runtime.misc
- RandomData in org.apache.commons.math3.random
- RandomDataImpl in org.apache.commons.math3.random

mi-hol, 2023-07-17:
> * NotNull in org.antlr.v4.runtime.misc

Reason for this seems usage of ANTLR (ANother Tool for Language Recognition) V 4.5 released Jan 23, 2015.
The current version is V  [4.13.0](https://github.com/antlr/antlr4/releases/tag/4.13.0) released May 21, 2023

Affected files are:

/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserBaseListener.java
/java_console/ui/src/main/java/com/rusefi/livedata/generated/CPP14ParserListener.java

they are both:

// Generated from CPP14Parser.g4 by ANTLR 4.5

unclear yet how to fix this issue because the only content of readme.md is "See https://github.com/antlr/grammars-v4/tree/master/cpp"
Would anyone have a hint?

mi-hol, 2023-07-17:
> * RandomData in org.apache.commons.math3.random
> * RandomDataImpl in org.apache.commons.math3.random

Reason seems that source files from library org.apache.commons.math3.random have been directly copied into RE source tree

mck1117, 2023-07-18:
Apache commons math lib fixed in 67decbdd4cf78447f26c5da42fd7cd379b601079

mi-hol, 2023-07-20:
now the new gradle seems to be the only source for deprecation warning :)

```
Configure project :ui
The AbstractArchiveTask.destinationDir property has been deprecated. This is scheduled to be removed in Gradle 8.0. Please use the destinationDirectory property instead. See https://docs.gradle.org/7.5.1/dsl/org.gradle.api.tasks.bundling.AbstractArchiveTask.html#org.gradle.api.tasks.bundling.AbstractArchiveTask:destinationDir for more details.
	at build_arh196x3twauwl53158hq[23](https://github.com/FOME-Tech/fome-fw/actions/runs/5607176588/jobs/10258082196#step:7:24)la$_run_closure3.doCall(/home/runner/work/fome-fw/fome-fw/java_console/ui/build.gradle:28)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
```

mck1117, 2023-07-20:
> now the new gradle seems to be the only source for deprecation warning :)

run `./gradlew clean` then build and you'll see all the warnings for every file

mi-hol, 2023-07-20:
> run `./gradlew clean` then build and you'll see all the warnings for every file

I lack the permission to perform this on the level org FOME builds, should this be added to the workflow definition?

mck1117, 2023-07-20:
all github actions happen in a fresh container - you only need that if building locally and have stale build intermediates

mck1117, 2023-07-20:
that's strange, I get warnings in a local build, but can't see them in the GHA builds

[image]

mi-hol, 2023-07-20:
correct, I looked up [this](https://github.com/FOME-Tech/fome-fw/actions/runs/5607176588/jobs/10258082196#step:7:24)  official  FOME build (not local build) and saw no warnings

mck1117, 2023-07-20:
by the way, the same builds that run here are running in your fork: https://github.com/mi-hol/fome-fw/actions

mi-hol, 2023-07-20:
> he same builds that run here are running in your fork

yes, I enabled GHA in my fork to test workflow changes before opening a PR. Do you see issues with that?

mck1117, 2023-07-20:
Nope, that's completely fine, and in fact recommended. Your builds are completely separate from FOME, don't have access to FOME build secrets, etc.

mi-hol, 2023-07-20:
> strange, I get warnings in a local build, but can't see them in the GHA builds

just to be sure, will you look into fixing this or shall I learn gradle?

mi-hol, 2023-07-27:
> Apache commons math lib fixed in [67decbd](https://github.com/FOME-Tech/fome-fw/commit/67decbdd4cf78447f26c5da42fd7cd379b601079)

@mck1117 as I'm neither a JAVA nor Gradle expert, I wonder if  maybe the same approach would work for [com.devexperts.util](https://docs.dxfeed.com/dxlib/api/com/devexperts/util/package-summary.html) ?
It seems a clone of this code exists in fome-fw/java_console/logging-api/src/main/java/com/devexperts/util/

mi-hol, 2023-07-28:
newest version of com.devexperts.logging.Logging is maintained on https://github.com/devexperts/QD/tree/master/dxlib

mi-hol, 2025-01-06:
fixed, no more search hits for https://github.com/FOME-Tech/fome-fw/actions?query=workflow%3A%22%22FOME+console%22%22+depreca

Grandle 9 obsoletes again something, see https://github.com/FOME-Tech/fome-fw/actions/runs/12606851461/job/35137498291

closed 2025-01-06 by mi-hol