FOME-Tech/fome-fw #219: Bench test improvements

mck1117, 2023-07-31:
- add button for "abort test" (the scenario is that your injector test bucket is about to overflow or you started a longer test than you want)
- increase maximum count
- independent parameters for fuel/spark/etc

125jdavis, 2023-07-31:
- Fuel pressure gauge in "Injector test" window in TS would be useful.
- Add "Fuel pump on" button in injector test window so that fuel pump can be kept on via ECU during injector tests.

125jdavis, 2023-07-31:
- Injector test window does not specify which injector(s) are being tested. Do we want to actuate all injectors? I'm not sure if I have an opinion yet.
- Also, there is no "test" button, only "Burn".

mck1117, 2023-08-03:
> Injector test window

Which injector test window? Screenshot?

125jdavis, 2023-08-03:
[image]

mck1117, 2023-08-10:
Used the new UI to flow test my fuel injectors, worked great!

closed 2023-08-25 by mck1117