FOME-Tech/fome-fw #222: Simulator uses 100% CPU

karniv00l, 2023-08-11:
Bug or a feature?

mi-hol, 2023-08-11:
@karniv00l did you look for similar reports in https://github.com/rusefi/rusefi/issues ?
I never got simulator to work on Windows 11 see https://github.com/rusefi/rusefi/issues/4817
Are you seeing 100% CPU on Linux, real hardware oo VM?

mi-hol, 2023-08-11:
> never got simulator to work on Windows 11 using virtual serial port

recommended workaround was to switch to a payed version of TunerStudio.

karniv00l, 2023-08-11:
[docker on mac](https://github.com/hyper-tuner/fome-simulator-docker) and ubuntu vps, I don't have a bare metal linux atm
however it's more like 80% CPU on a single core vps

mi-hol, 2023-08-11:
@karniv00l suggest to update issue title with above info

mck1117, 2023-08-11:
I believe that's by design. The "idle thread" sits and spin waits to "serve interrupts", so you're going to get high cpu usage.

mck1117, 2023-08-11:
see `_sim_check_for_interrupts()`

mck1117, 2023-08-11:
and @mi-hol,  please try to keep the issue on-topic. If you're having other simulator issues or questions, make a new issue. This one is about high CPU usage.

karniv00l, 2023-08-12:
Alright, maybe not worth keeping this open then

closed 2023-08-12 by karniv00l

mi-hol, 2023-08-12:
> @mi-hol, please try to keep the issue on-topic. If you're having other simulator issues or questions, make a new issue.

@mck1117 I thought the commonality with my issue was eventually usage of virtual serial port.

Regarding simulator for FOME it's unclear (at least to me) if that is actually a supported component because:
- FOME has no reference for simulator in wiki 
- previous comments from Andrey and you in the quoted issue suggested no interested to support it using  virtual serial port

Considering above, is it worth to create an issue in FOME?