FOME-Tech/fome-fw #235: Instances mentioning rusefi

diy-efi, 2023-09-05:
As a new user, I've been through a lot of the tunerstudio interface.
I mentioned on discord I saw some RE mentions, and a ticket seemed a good place to capture.

### TRIGGER

Trigger Type tooltip
[image]

Invert Primary
[image]

Invert Secondary
[image]

Cam inputs
[image]

### STATUS LEDS

Running status LED
[image]

 TS Comms LED
 
[image]

### IDLE Settings

Detailed status in consol
[image]

### Electronic throttle body

URL
[image]

Debug mode
[image]

### LUA Outputs

URL
[image]

### Also a few mentions of RUSEFI wideband, I guess that is correct.

[image]

[image]

mi-hol, 2023-09-12:
I verified the field
> Invert Secondary [image]

and above mentioned obsolete references are found in generated files i.e. "fome-fw/firmware/tunerstudio/generated/fome_mre_f4.ini"
`
invertSecondaryTriggerSignal = "https://wiki.rusefi.com/Trigger-Configuration-Guide\nThis setting flips the signal from the secondary engine speed sensor."
`
Instructions in this file how to change it:
```
; In TunerStudio some fields have little question mark on the left of the name for additional field tips.
; Those tips are defined in ../integration/rusefi_config.txt
```
@diy-efi Would you be in a position to fix the reported issues?

mi-hol, 2023-09-12:
**note**: 
URL links in tooltips should use "redirected" links defined in https://github.com/FOME-Tech/wiki/blob/master/docusaurus.config.js#L104C1-L104C1 to lower maintenance burden

mi-hol, 2023-09-12:
first fix in https://github.com/FOME-Tech/fome-fw/pull/242