FOME-Tech/fome-fw #237: 2JZ Trigger Issues - 1G-FE Engine

diy-efi, 2023-09-07:
I am currently prototyping an ECU for an IS200, which has a 1G-FE 6cyl engine.
It shares the same crank/cam pattern as the later 2JZ VVT engine.
A 36-2 wheel at crank speed, and a secondary cam signal of 3 evenly spaced teeth.

The factory wiring for this uses a shared ground for both VR sensors.
With 2 MAX9924 chips, this resulted in crosstalk on each waveform.

Each sensor is now separately wired, each with their own shielded cable.

The car starts and run fine on crank only.

However, when the 2JZ cam pattern is selected and require cam/vvt for ignition is selected, the engine fails to start and there are no injection/ignition events on state1-6 of each.

A logic analyser hooked up to the VR1/VR2 pins of the polygonus board shows a near perfect signal of what I would expect.

The tunerstudio trigger logger only shows 1 spike per 720, though the log file seems to show 3 teeth.
[image]

TS trigger log
[cam own loom.csv](https://github.com/FOME-Tech/fome-fw/files/12552565/cam.own.loom.csv)

logic analyser file
[1g-fe with cam wired.zip](https://github.com/FOME-Tech/fome-fw/files/12552567/1g-fe.with.cam.wired.zip)

Current tune - far from ideal, but starts and idles on crank only
[CurrentTune.zip](https://github.com/FOME-Tech/fome-fw/files/12552575/CurrentTune.zip)

diy-efi, 2023-09-11:
Looks like this could be the issue:  https://github.com/rusefi/rusefi/issues/5557 
The 2JZ decoder was never complete: https://rusefi.com/forum/viewtopic.php?f=4&t=2613&sid=98b794fa1ae824c455a2308ef96245cf

mck1117, 2023-09-14:
@diy-efi, give the latest snapshot a try. I've simplified it a bit so that you don't need extra parameters for sync to work. If you can grab a regular datalog (not trigger log) of cranking/running that'd be awesome.

diy-efi, 2023-09-14:
This solved it for me, thanks!

A few things I noticed.
1. The FOME console app in the snapshot would not load via auto or manual DFU, I ended up copying the binary file to the last release and using that console and it worked, not sure if thsi is expected at the moment?
2. Before my trigger angle was 170, to get a VVT angle of 0, the VVT offset would have been ~-10. It seemed unhappy with that, so I had to set my base trigger + a cycle (530) and ended up with a VVT angle of 350

Logs...
[1gfe_snapshot_logs.zip](https://github.com/FOME-Tech/fome-fw/files/12611232/1gfe_snapshot_logs.zip)

diy-efi, 2023-09-14:
One potential suggestion before I close this.
What do you think on renaming to "Toyota Tri-Lobe" or simmilar, as it's not just a 2JZ decoder, just that's most popular.
Off the top of my head it's 1JZ, 2GZ, 1G-FE (and also a 3G something...)

mck1117, 2023-09-14:
renamed to "toyota 3 tooth" in 4ddd1ffbdb8bd0c8b7f7eb0e4ca30997035d00f0

diy-efi, 2023-09-15:
Great stuff, thanks for your help on this one 👍

closed 2023-09-15 by diy-efi