FOME-Tech/fome-fw #243: GHA: HW-CI failing with "timeout" frequently

mi-hol, 2023-09-12:
Examples:
https://github.com/FOME-Tech/fome-fw/actions/runs/6157295373/job/16707734966?pr=242
https://github.com/FOME-Tech/fome-fw/actions/runs/6156325871/job/16704769950

mi-hol, 2023-09-12:
```
I 230912 074536.712 [main] IoUtil - Command [reset_engine_chart] executed in 4950
I 230912 074536.761 [communication executor1] IncomingDataBuffer - output channels header: timeout 5000ms. Got only 0 byte(s) while expecting 2
I 230912 074541.761 [communication executor1] IncomingDataBuffer - text header: timeout 5000ms. Got only 0 byte(s) while expecting 2
E 230912 074541.761 [communication executor1] BinaryProtocol - ERROR: TS_GET_TEXT failed
E 230912 074541.762 [Commands Queue] BinaryProtocol - timeout sending [reset_engine_chart] giving up: java.util.concurrent.TimeoutException
E 230912 074546.713 [Commands Queue] CommandQueue - Major connectivity error
java.lang.IllegalStateException: No confirmation received after timeout of 4950 ms
	at com.rusefi.io.CommandQueue.sendCommand(CommandQueue.java:99)
	at com.rusefi.io.CommandQueue.sendPendingCommand(CommandQueue.java:65)
	at com.rusefi.io.CommandQueue.access$000(CommandQueue.java:26)
	at com.rusefi.io.CommandQueue$1.run(CommandQueue.java:114)
	at java.base/java.lang.Thread.run(Thread.java:829)
E 230912 074546.715 [Commands Queue] IoUtils - CommandQueue error: java.lang.IllegalStateException: No confirmation received after timeout of 4950 ms
java.lang.Exception
	at com.rusefi.util.IoUtils.exit(IoUtils.java:16)
	at com.rusefi.io.CommandQueue.lambda$static$1(CommandQueue.java:103)
	at com.rusefi.io.CommandQueue$1.run(CommandQueue.java:117)
	at java.base/java.lang.Thread.run(Thread.java:829)
Error: Process completed with exit code 254.
```

mck1117, 2023-09-25:
is it always the same spot in the hw-ci job that it fails? is it always this one command?

mi-hol, 2024-02-13:
@mck1117 issue seems fixed 
used https://github.com/FOME-Tech/fome-fw/actions/runs/7869816668/job/21469703191 to verify
```
[1760](https://github.com/FOME-Tech/fome-fw/actions/runs/7869816668/job/21469703191#step:10:1761)
I 240212 094419.431 [main] IoUtil - AUTOTEST RPM change [2000] executed in 254
I 240212 094419.431 [main] IoUtil - Sending command [set ignition_mode 2]
I 240212 094419.431 [Commands Queue] BinaryProtocol - Sending [set ignition_mode 2]
I 240212 094419.537 [Commands Queue] IoUtil - Got confirmation in 106ms
I 240212 094419.537 [main] IoUtil - Command [set ignition_mode 2] executed in 106
I 240212 094419.537 [main] IoUtil - Sending command [reset_engine_chart]
I 240212 094419.537 [Commands Queue] BinaryProtocol - Sending [reset_engine_chart]
I 240212 094419.539 [Commands Queue] IoUtil - Got confirmation in 2ms
I 240212 094419.539 [main] IoUtil - Command [reset_engine_chart] executed in 2
I 240212 094419.633 [communication executor1] MessagesCentral - postMessage EngineState: AUX Serial started
I 240212 094419.633 [communication executor1] MessagesCentral - postMessage EngineState: Emulating position sensor(s). RPM=2000
I 240212 094419.633 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_set ignition_mode 2:19
I 240212 094419.633 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_reset_engine_chart:18
I 240212 094419.633 [communication executor1] MessagesCentral - postMessage EngineState: Length should be positive
I 240212 094420.441 [main] TestingUtils - got next chart in 902ms for engine_type 3
I 240212 094420.446 [main] IoUtil - Sending command [set ignition_mode 1]
I 240212 094420.446 [Commands Queue] BinaryProtocol - Sending [set ignition_mode 1]
I 240212 094420.451 [Commands Queue] IoUtil - Got confirmation in 5ms
I 240212 094420.451 [main] IoUtil - Command [set ignition_mode 1] executed in 5
I 240212 094420.451 [main] IoUtil - Sending command [reset_engine_chart]
I 240212 094420.451 [Commands Queue] BinaryProtocol - Sending [reset_engine_chart]
I 240212 094420.453 [Commands Queue] IoUtil - Got confirmation in 2ms
I 240212 094420.453 [main] IoUtil - Command [reset_engine_chart] executed in 2
I 240212 094420.536 [communication executor1] MessagesCentral - postMessage EngineState: AUX Serial started
I 240212 094420.536 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_set ignition_mode 1:19
I 240212 094420.536 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_reset_engine_chart:18
```

closed 2024-02-13 by mi-hol