FOME-Tech/fome-fw #244: Should TS "Debug mode" be removed?

mi-hol, 2023-09-12:
@nmschulte you had made this suggestion in https://github.com/rusefi/rusefi/issues/3614#issuecomment-1354347339 and I stumbled over it today only after I had referenced  the empty FOME wiki page in https://github.com/FOME-Tech/fome-fw/pull/242

Andrey never answered on questions about this question.
I wonder if we can get to a decision within FOME-Tech team.
Any comments?

nmschulte, 2023-09-12:
I think this was addressed here; resolution was to keep as it is useful to those that know how, despite any performance/resource concerns: https://github.com/FOME-Tech/FOME-Baseline/issues/8

mi-hol, 2023-09-13:
> https://github.com/FOME-Tech/FOME-Baseline/issues/8#issuecomment-1373069122): https://github.com/FOME-Tech/FOME-Baseline/issues/8

seems a private repo, I don't have access

mi-hol, 2023-09-13:
with the decision taken to keep debug mode the currently empty FOME wiki page needs content.

is the RE content copied below applicable?
"Debug fields is an advanced troubleshooting feature allowing one to monitor the internal state of some rusEFI subsystems.

In TunerStudio, select "Base Engine Settings -> Debug Mode" and use the gauges from "Debug" category.

See [Debug Fields](https://wiki.rusefi.com/Debug-Fields/)"

nmschulte, 2023-09-13:
> > [FOME-Tech/FOME-Baseline#8 (comment)](https://github.com/FOME-Tech/FOME-Baseline/issues/8#issuecomment-1373069122)): [FOME-Tech/FOME-Baseline#8](https://github.com/FOME-Tech/FOME-Baseline/issues/8)
> 
> seems a private repo, I don't have access

Oh, my bad.  I didn't realize this was a different repo.

mi-hol, 2023-09-13:
> more worthwhile to just move the stuff we care about to their own channels and remove the debug modes

@mck1117 sounds great, but will this task make it into the list of changes to address in one of the next iterations/releases?
There is currently no public visibility about priorities (i.e. via labels on issues or project with assigned issues)

OrchardPerformance, 2023-09-16:
@nmschulte Nothing sinister about it, just an old repo we used initially to figure out where we were at. Left it private as we are slowly obsoleting it and didn't want it to cause any confusion with the main repo.
All the tickets there will eventually be moved over here or closed then the repo closed.

closed 2025-01-06 by mi-hol