FOME-Tech/fome-fw #246: Knock retard not activating (20230516 release)

OrchardPerformance, 2023-09-16:
Based on the testing we did at silverstone it seems the knock retard is not activating, knock count increases, and the volume on the sensors passes the threshold but no timing is removed. 

Over the day we tried several combinations of aggression (5 and 10%), threshold (lowered it by 2dBv) and max allowed retard (tried several tables then flat mapped) with no change in result. 

Logs and tune attached.  
[re_17.zip](https://github.com/FOME-Tech/fome-fw/files/12640948/re_17.zip)

OrchardPerformance, 2023-09-16:
[TunerStudioDebugPackage (1).zip](https://github.com/FOME-Tech/fome-fw/files/12640961/TunerStudioDebugPackage.1.zip)

mck1117, 2023-09-25:
OK, so I had a look at this, and while knock count _is_ incrementing as expected, I'm really not sure why it's not retarding timing.

I've improved logging a bit in #261, give it another shot if you can once that's merged.

OrchardPerformance, 2023-09-25:
Thanks, I did think it was odd, currently putting a matching setup on my e30 to try it again. 

Guess we proceed to the next release and push out a hotfix if there is an issue? 
Or do you want me to put some miles on the release candidate to see if we can find it before release?

mck1117, 2023-10-04:
@OrchardPerformance please try to put some miles on a snapshot first

OrchardPerformance, 2023-10-04:
Will do

OrchardPerformance, 2023-11-05:
Catching up with this one, we ran snapshot https://github.com/FOME-Tech/fome-fw/actions/runs/6580882353

Result was still no response from the knock even though it was counting upwards. 
Almost like the max allowed retard was fixed at zero. 

Attached a log, the FW and the tune if anyone wants to check it further. 
[FW and Tune.zip](https://github.com/FOME-Tech/fome-fw/files/13260383/FW.and.Tune.zip)

[re_56_went_bang.zip](https://github.com/FOME-Tech/fome-fw/files/13260374/re_56_went_bang.zip)

I'll keep looking into this when I have time, now have the knock sensors working on the road car so will try to get a baseline from that.

nmschulte, 2024-03-10:
I modified the simulator (`master`) to produce knock signal, also shows the logic is working.  However, simulator is single-threaded (IIUC), and my hacking doesn't account for real-world timings

```diff
diff --git a/firmware/controllers/engine_cycle/knock_controller.cpp b/firmware/controllers/engine_cycle/knock_controller.cpp
index eef94df816..c0fbf0e63a 100644
--- a/firmware/controllers/engine_cycle/knock_controller.cpp
+++ b/firmware/controllers/engine_cycle/knock_controller.cpp
@@ -130,6 +130,7 @@ __attribute__((weak)) void onStartKnockSampling(uint8_t cylinderNumber, float sa
 	UNUSED(cylinderNumber);
 	UNUSED(samplingTimeSeconds);
 	UNUSED(channelIdx);
+	engine->module<KnockController>()->onKnockSenseCompleted(cylinderNumber, -30, getTimeNowNt());
 }
 
 static uint8_t cylinderNumberCopy;
@@ -154,10 +155,6 @@ static void startKnockSampling(void*) {
 void Engine::onSparkFireKnockSense(uint8_t cylinderNumber, efitick_t nowNt) {
 	cylinderNumberCopy = cylinderNumber;
 
-#if EFI_SOFTWARE_KNOCK
 	scheduleByAngle(nullptr, nowNt,
 			/*angle*/engineConfiguration->knockDetectionWindowStart, startKnockSampling);
-#else
-	UNUSED(nowNt);
-#endif
 }
```

[image]

---

Just thinking aloud here:
- `onFastCallback` period is 5 ms
- Looking at the last tune and log provided: knock happening w/: 6 cyl, ~3000 RPM, ~10° max retard, 60° sample duration (limited by firmware to 2000 samples @ sample rate (mck; know this for Proteus?))
  - `3000 RPM / 2 * 6 = 9000 spark-per-minute = 150 spark-per-second => 6.66 ms sample period`

This shouldn't be a case of re-apply squelching knock-retard, I don't think, at least not at 3k RPM.
Is this something about threading?  Should I modify simulator to properly thread, assuming it doesn't?

PR #419 (2024-05-04): fix knock retard table init
column and row definition were swapped w/re: value lookup

fixes #246, #417

----

Andrey was right: https://github.com/FOME-Tech/fome-fw/issues/417#issuecomment-2087816779
> is this https://github.com/rusefi/rusefi/issues/6370?

Comparison to boost; RPM is columns:
[image]

boost:
```cxx
// init
boostMapOpen.init(config->boostTableOpenLoop, config->boostTpsBins, config->boostRpmBins);
boostMapClosed.init(config->boostTableClosedLoop, config->boostTpsBins, config->boostRpmBins);

// getValue
float openLoop = luaOpenLoopAdd + m_openLoopMap->getValue(rpm, driverIntent [... cut]
commit b8d7b2d7 (2024-05-04): fix knock retard table init (#419)

* fix knock retard table init

column and row definition were swapped w/re: value lookup

fixes #246, #417

* initialize knock controller / max retard table on reset

* fix KnockController init

no need for previousConfig (or any config ref); use the current config
closed 2024-05-04 by mck1117