FOME-Tech/fome-fw #248: Multi-dimensional mapping improvements

125jdavis, 2023-09-17:
Currently we have adder tables for VE, spark, and boost tables. In many cases a multiplying or blending (interpolation) would be the better and/or more intuitive approach, so allowing the user to select add/multiply/blend allows more flexibility.

Shown below is one potential approach for blending: 
[image]

Note that the axes of the authority table need not necessarily match the axes of the primary and secondary maps.

OrchardPerformance, 2023-09-17:
Looks like it, thanks. 

How hard is it to put a switch in that allows the adder table to either be an adder, or makes table 1 a full VE table? 
Or just make table one a full VE table and only 2-4 adders to satisfy those that want a full table switch, given it comes up a LOT.

125jdavis, 2024-01-15:
I like the switch approach. Additionally, one of the weaknesses of the current approach is that the bias parameter and the y-axis of the blend table must be the same. There are cases where they should not be the same.

mck1117, 2024-01-15:
>  bias parameter and the y-axis of the blend table must be the same

This is...not true. The Y axis of the blend table is always the same as the Y axis of the table its blending (VE blends use VE table Y axis, etc), and the blend parameter is of course selectable.

125jdavis, 2024-01-17:
Understood. So in order to make a switchable axis algorithm (i.e. SD at low load, TPS at high load) the y-axis of the blend table would need to be selectable.

OrchardPerformance, 2024-01-17:
Yep, blend table axis should probably be selectable. 
Cases where it is needed are rare but it gives a huge gain in function.

The 1st blend curve should also probably be a 8x8 table instead of a curve so that first one can be used as a blend map with rpm acounted for.

mck1117, 2024-01-17:
You can now pick many parameters as the Y axis: cde642ae6180e1e82664b08e5c0d7956ce53a8ef