FOME-Tech/fome-fw #250: Additional Channels for Logging

OrchardPerformance, 2023-09-17:
Ticket to list the extra channels people need to get better data logging done on their race cars. 

Oil temp 
Fuel temp - Comes from the flex sensor 
Ambient temp -

diy-efi, 2023-09-17:
For the fuel temp consider allowing the user to have an option to pull from flex sensor or dedicated analogue input.
Using the Bosch analogue combined temp/pressure sensor is becoming more popular.

One each on fuel and oil is easy data.

OrchardPerformance, 2023-09-17:
There are some arguments for ambient pressure, pre-intercooler and post intercooler pressure (with temperatures). 

For anyone adding a turbo or tracking a turbo car they can be used to determine intercooler efficiency and if they are having heat soak issues

125jdavis, 2023-09-17:
For those wanting traction control (i'm hoping to make it in LUA eventually)
Front VSS
Rear VSS

or one for each corner

125jdavis, 2023-09-17:
EMAP (exhaust pressure)

mck1117, 2023-09-18:
#254 related

OrchardPerformance, 2023-09-20:
"Flow Rate" for logging EWP or charge cooler or oil system fluid flow?