FOME-Tech/fome-fw #251: /firmware/tunerstudio /rusefi.input has many misleading hints due to CopyPaste and missed changes

mi-hol, 2023-09-17:
One specific example in https://discord.com/channels/1060875162892898324/1100970095385268295/1153052695209451531

125jdavis, 2023-09-17:
For clarity's sake, maybe the VE blend tables should be called trim tables instead of adder tables.

mck1117, 2023-09-18:
Let's please make specific GH issues for a specific problem to be fixed. A broad issue like this will never be closed. The VE blend issue has been moved to #252.

closed 2023-09-18 by mck1117

mi-hol, 2023-09-18:
> Let's please make specific GH issues for a specific problem to be fixed. A broad issue like this will never be closed

@mck1117 please allow me to have a different view.
This a case where a "bigger picture" is required first to reach a common understand of the problem at hand.
Then several derived actions will result in specific issues to fix one at a time.

How do you suggest to address such a large issue instead?