FOME-Tech/fome-fw #256: Add timing and VE gauge to VE and Ignition table

OrchardPerformance, 2023-09-20:
Adding a couple of gauges to the VE and Ignition table would help anyone tuning to monitor the other critical variables while looking at the tuning tables on a small screen. 

Suggest adding Ignition Timing and Lambda to the VE table window and VE and Lambda to the Ignition timing table window. 

Thinking of something like the TPS sensor menu or the mock up below. 

[image]

125jdavis, 2023-09-22:
Is it possible to have AFR/lambda as a line graph? I find those more useful than a round gauge.

OrchardPerformance, 2023-09-25:
I don't see why not, probably more compact to have both like that too. 

I'm used to using both at once so no care either way.