FOME-Tech/fome-fw #257: GM map sensor calibrations

OrchardPerformance, 2023-09-20:
I am not sure our GM default calibrations are correct, the 3 bar we had was 10kpa out at ambient on the dyno. 

Using the calibrations below corrected it.

OrchardPerformance, 2023-09-20:
[image]

mck1117, 2023-09-25:
The numbers we have in the firmware (for the 3 bar, at least) agree exactly with this: https://www.bmotorsports.com/shop/product_info.php/cPath/0_129_143/products_id/1584

OrchardPerformance, 2023-11-05:
hmmm, 2 competing calibrations for what should be the same sensor, both correct in the context of their specific product. 

Put this one on the back burner and I will look into why these 2 versions of the same sensor differ. No wonder people complain about the variability/reliability of these GM sensors.