FOME-Tech/fome-fw #289: ignTrims vs timing_offset_cylinder

rusefillc, 2023-11-02:
ignTrims is used by advance_map.cpp and timing_offset_cylinder used in engine_math.cpp

Are both of these still needed? If both are needed are those similar enough to be handled closer to each other in some sort of getCombinedCylinderIgnitionTrim method?

mck1117, 2024-01-17:
These serve two fundamentally different functions.

timing_offset_cylinder = "I have a funny engine with cylinders not evenly spaced every 720 / N degrees". Offsets both ignition and fueling.

ignTrims = "I have an engine that needs slightly more/less ignition timing on some cylinder under different conditions for optimal power" Offsets just ignition, but as a function of load/rpm, instead of fixed.

closed 2024-01-17 by mck1117

rusefillc, 2024-01-17:
``timing_offset_cylinder;per-cylinder timing correction`` vs ``Offsets both ignition and fueling.`` sounds like some rename/tootltip opportunity

rusefillc, 2024-01-17:
@mck1117 should the tooltip be ``per-cylinder ignition and fueling timing correction for uneven engines``?