FOME-Tech/fome-fw #295: Leverage unused variable detection mechanism

rusefillc, 2023-11-03:
this repo has at least one (very minor) defect which would have been detected by unused-variable error had it not be disabled

https://github.com/rusefi/rusefi/commit/e78803b314a4829e58b6c911067a4e611ee1f4f0

mck1117, 2023-11-03:
what's the bug? why do you make a passive aggressive ticket instead of helping with the issue at hand?

closed 2023-11-03 by mck1117

rusefillc, 2023-11-03:
Huh? You were tagged on https://github.com/rusefi/rusefi/commit/a07f82404a38d731a979498315a2771bb251ee62 yesterday 

This issue is not about the bug but about leveraging tools which help us create better software. You know better than closing.

mck1117, 2023-11-03:
That was fixed 8 months ago: d36dfe4e07e6c00c26a3365fbe1bbabb0bbacbb0

Try and make issues useful - if there's a known fix, link the relevant commit/code/whatever instead of a cryptic warning..