FOME-Tech/fome-fw #321: DFU updater does not work on Windows

rusefillc, 2023-12-25:
```
Console version 20230220
Windows 10.0
Bundle fome.release_2312.f407-discovery
Executing wmic path win32_pnpentity where "Caption like '%STM32%' and Caption like '%Bootloader%'" get Caption,ConfigManagerErrorCode /format:list

Caption=STM32  BOOTLOADER

ConfigManagerErrorCode=0

Caption=STM32  BOOTLOADERConfigManagerErrorCode=0

Executing .\STM32_Programmer_CLI/bin/STM32_Programmer_CLI.exe -c port=usb1 -w "C:\stuff\bundles\fome_f407-discovery\fome.release_2312.f407-discovery\console\..\fome.bin" 0x08000000 -v -s
ERROR: does not look like DFU has worked!
Devices: OK         USBDevice       STM32  BOOTLOADER                                                                USB\VID_...
0 / 0
hint: error state is already in your clipboard, please use PASTE or Ctrl-V while reporting issues

```

rusefillc, 2023-12-25:
[image]

rusefillc, 2023-12-25:
https://github.com/rusefi/rusefi/commit/697f799b9e95d80bb7e4384302988d974c15d2a1

mck1117, 2023-12-25:
fixed in https://github.com/FOME-Tech/fome-fw/pull/324

closed 2023-12-25 by mck1117