FOME-Tech/fome-fw #343: JAVA17 compiling console fails with 17 errors

mi-hol, 2024-01-09:
https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887
```
./gradlew test
  shell: /usr/bin/bash -e {0}
  env:
    JAVA_HOME: /opt/hostedtoolcache/Java_Zulu_jdk/17.0.9-9/x64
    JAVA_HOME_17_X64: /opt/hostedtoolcache/Java_Zulu_jdk/17.0.9-9/x64

> Configure project :autotest
The Project.getConvention() method has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#deprecated_access_to_conventions
The org.gradle.api.plugins.Convention type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#deprecated_access_to_conventions

> Configure project :config_definition
The org.gradle.util.ConfigureUtil type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#org_gradle_util_reports_deprecations
	at build_6j4a4kqa6kto67udrkvhpe94f.run(/home/runner/work/fome-fw/fome-fw/java_tools/configuration_definition/build.gradle:18)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
The org.gradle.api.plugins.JavaPluginConvention type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#java_convention_deprecation
	at build_6j4a4kqa6kto67udrkvhpe94f.run(/home/runner/work/fome-fw/fome-fw/java_tools/configuration_definition/build.gradle:18)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)

> Configure project :enum_to_string
Build file '/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle': line 20
The org.gradle.api.plugins.BasePluginConvention type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#base_convention_deprecation
	at build_7ryml424g2mrufz6bete7q6gy$_run_closure2.doCall$original(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:20)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
	at build_7ryml424g2mrufz6bete7q6gy.run(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:19)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
Build file '/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle': line 20
The org.gradle.api.plugins.Convention type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.4/userguide/upgrading_version_8.html#deprecated_access_to_conventions
	at build_7ryml424g2mrufz6bete7q6gy$_run_closure2.doCall$original(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:20)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
	at build_7ryml424g2mrufz6bete7q6gy.run(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:19)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
Build file '/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle': line [3](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:3)9
The org.gradle.util.ConfigureUtil type has been deprecated. This is scheduled to be removed in Gradle 9.0. Consult the upgrading guide for further information: https://docs.gradle.org/8.[4](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:4)/userguide/upgrading_version_8.html#org_gradle_util_reports_deprecations
	at build_7ryml424g2mrufz6bete7q6gy$_run_closure3$_closure[5](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:5).doCall$original(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:39)
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)
	at build_7ryml424g2mrufz[6](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:6)bete[7](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:8)q6gy$_run_closure3.doCall$original(/home/runner/work/fome-fw/fome-fw/java_tools/enum_to_string/build.gradle:3[8](https://github.com/FOME-Tech/fome-fw/actions/runs/7462204158/job/20303988887#step:9:9))
	(Run with --stacktrace to get the full stack trace of this deprecation warning.)

> Task :core_io:compileJava
> Task :core_io:processResources NO-SOURCE
> Task :core_io:classes
> Task :core_io:jar

> Task :logging-api:compileJava
/home/runner/work/fome-fw/fome-fw/java_console/logging-api/src/main/java/com/devexperts/util/IndexedSet.java:483: warning: [dep-ann] deprecated item is not annotated with @Deprecated

	public boolean contains(Object value) {
	               ^
/home/runner/work/fome-fw/fome-fw/java_console/logging-api/src/main/java/com/devexperts/util/IndexerFunction.java:346: warning: [deprecation] Indexer in com.devexperts.util has been deprecated
			return Indexer.DEFAULT;
			       ^
/home/runner/work/fome-fw/fome-fw/java_console/logging-api/src/main/java/com/devexperts/logging/Logging.java:187: warning: [deprecation] newInstance() in Class has been deprecated
				impl = (DefaultLogging)Class.forName(className).newInstance();
				                                               ^
  where T is a type-variable:
    T extends Object declared in class Class
/home/runner/work/fome-fw/fome-fw/java_console/logging-api/src/main/java/com/devexperts/logging/Logging.java:214: warning: [deprecation] newInstance() in Class has been deprecated
				impl = (DefaultLogging)Class.forName("com.devexperts.logging.Log4j2Logging").newInstance();
				                                                                            ^
  where T is a type-variable:
    T extends Object declared in class Class
4 warnings
> Task :logging-api:processResources NO-SOURCE
> Task :logging-api:classes
> Task :logging-api:jar
/home/runner/work/fome-fw/fome-fw/java_console/inifile/src/main/java/com/rusefi/tune/xml/Constant.java:3: error: package javax.xml.bind.annotation does not exist
import javax.xml.bind.annotation.XmlAttribute;
                                ^
[... 63 lines cut ...]
@XmlSchema(
 ^
  symbol: class XmlSchema
/home/runner/work/fome-fw/fome-fw/java_console/inifile/src/main/java/com/rusefi/tune/xml/package-info.java:3: error: cannot find symbol
        elementFormDefault = XmlNsForm.QUALIFIED
                             ^
  symbol: variable XmlNsForm
17 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':inifile:compileJava'.

> Task :inifile:compileJava FAILED
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --info option to get more log output.
> Run with --scan to get full insights.

BUILD FAILED in 3s
5 actionable tasks: 5 executed
Error: Process completed with exit code 1.
```

mi-hol, 2024-01-09:
https://github.com/GoogleChromeLabs/bubblewrap/issues/791#issuecomment-1589077234 seems to indicate an outdated Android SDK  as root cause

mi-hol, 2024-01-14:
updating java_console/inifile/src/main/java/com/rusefi/tune/xml/package-info.java
to comment below reduces the number of errors from 17 to 15

```
// @XmlSchema(
//        namespace = "http://www.msefi.com/:msq",
//        elementFormDefault = XmlNsForm.QUALIFIED
// )

mi-hol, 2024-01-14:
java_console/.idea/misc.xml

```
  <component name="ProjectRootManager" version="2" languageLevel="JDK_1_8" default="false" project-jdk-name="1.8" project-jdk-type="JavaSDK">
```
and

java_console/.idea/libraries/XML_for_11.xml  seems to set jaxb library version

```
<component name="libraryTable">
  <library name="XML for 11">
    <CLASSES>
      <root url="jar://$PROJECT_DIR$/lib/jaxb-api.jar!/" />
    </CLASSES>
    <JAVADOC />
    <SOURCES />
  </library>
</component>
```

mi-hol, 2024-01-15:
from https://javaee.github.io/jaxb-v2/doc/user-guide/ch06.html#d0e6862
"
Q: | Can I run my existing JAXB 1.0.x applications on the JAXB 2.0 runtime?
A: | This is no longer supported. However, you should be able to deploy http://search.maven.org/remotecontent?filepath=com/sun/xml/bind/jaxb1-impl/2.2.5-1/jaxb1-impl-2.2.5-1.jar with your with your application application.
"

mi-hol, 2024-01-15:
updated java_console/ui/build.gradle with `apply from: '../../java_tools/dependencies.gradle`
but java_tools/jaxb1-impl-2.2.5-1.jar gets NOT used for compiling java_console/ui/src/main/java/com/rusefi/tools/ConsoleTools.java as the other java source files in java_console/inifile/src/main/java/com/rusefi/tune/xml/

What do I miss?

mi-hol, 2024-01-16:
Thanks to Andrey's hint pointing to https://github.com/rusefi/rusefi/commit/513b075eae1300b2bae0ca6c4656960f35300646 everything compiles now successfully

mck1117, 2024-01-19:
> Thanks to Andrey's hint pointing to [rusefi/rusefi@513b075](https://github.com/rusefi/rusefi/commit/513b075eae1300b2bae0ca6c4656960f35300646) everything compiles now successfully

PR?

mi-hol, 2024-01-19:
> PR?

Would love to cherry-pick but it seems I screwed up my local fork when I used git commands in pwsh :(
GH Desktop seems limited to work only with a single repo, did I get this right?

closed 2025-01-06 by mi-hol