FOME-Tech/fome-fw #352: Reset PID values after engine shutoff

125jdavis, 2024-01-25:
[image]

Had a failed start. Idle PID tried to save it, but failed. I term went really high and stayed there, so when I restarted, there was a horrible RPM hunting because the I term was still stuck high.

Idle PID terms should reset after engine restart. Maybe all PID (boost, ETB, Alternator, WG, etc) terms should reset unless there is a strong case not to.

mck1117, 2024-01-29:
Can you send that log? I want to peek at a few things in there.

I think we already try to reset PID when not running, but I'm not sure what's up in your log.

125jdavis, 2024-02-14:
https://hypertuner.cloud/#/t/4OMw3o1W7_/logs/2024_01_24_13_36_58_idle_cut_TOXp4TNenA.csv

Sorry for the delay, hope this works.

125jdavis, 2024-02-14:
> I think we already try to reset PID when not running, but I'm not sure what's up in your log.

It could be that I never keyed off after the engine died. I'm not sure if PID values reset when RPM == 0.

NoTwixD, 2025-06-15:
I had similar issue with idle PID , Idle was poorly tuned, when car died on its own, I term wind up hard , and when i cranked car again the rpm was higer than idle rpm limit , so we were unable to lower idle without reseting ecu

NoTwixD, 2026-05-15:
I think this issue was solved , with lastest firmware , i do not observed it last time 

[image]

car from log above was stall with clutch by mistake , at first iterm rised , and than i term reseted itself back to 0 so in my opinion works as intended