FOME-Tech/fome-fw #379: knock frequency vs. bore confusion

mck1117, 2024-03-10:
The tooltips are completely wrong about what these two settings do

nmschulte, 2024-03-10:
It seems `cylinderBore` isn't ever referenced in fome-fw (except to set default and per-board-config overrides); shall remove or?

mck1117, 2024-03-10:
I see a few different fixes here
- remove `cylinderBore`, fix tooltips, don't change behavior
- calculate correct frequency from `cylinderBore`, unless the frequency is set, then use that (do what the tooltips say)
- calculate correct frequency from `cylinderBore`, delete frequency

OrchardPerformance, 2024-04-07:
I think the short term fix may be to remove the cylinder bore from TS while we work out a better fix. 

Bench testing has shown the control working right as long as the bore is not set. 
We can link people to a frequency calculator via the TStip.

We need the frequency override to chose 1st or 2nd harmonic, though I guess that could be done by setting an "incorrect" bore size.

mck1117, 2024-08-06:
fixed in 7bf122f5e94112622d7a3fcb40a8917e62e6f222

closed 2024-08-06 by mck1117