FOME-Tech/fome-fw #382: toyota 3 tooth weirdness

mck1117, 2024-03-11:
> He's seeing an issue where sometimes it looks like the cam decoder is picking up on the wrong tooth.

> If he gets a high VVT angle (120+) as the ECU is powered, endless cranking won't change that.   If it's low (near 0) it'll start fine.

Here's the bad behavior:

https://github.com/FOME-Tech/fome-fw/assets/568254/cf85ad62-6b20-4a31-ae11-6bc12c4e3edc

[matches_vid.zip](https://github.com/FOME-Tech/fome-fw/files/14564563/matches_vid.zip)

Waiting on a snip of engine sniffer, and a log of good behavior.

mck1117, 2024-03-11:
Engine sniffer from the same car

[image]

mck1117, 2024-03-11:
This cam tooth is very close to the sync point

[image]

diy-efi, 2024-03-12:
[1g-fe with cam wired.zip](https://github.com/FOME-Tech/fome-fw/files/14578750/1g-fe.with.cam.wired.zip)

This is a salee log from the development car I had briefly.

Wiring 
E23 (crank +)  VR1+ (on max9926)
E22 (crank -) VR1- (on max9926)

E10 (cam +) VR2+  (on max9926)
New wire (cam -) VR2- on max9926)
E10 is also a new wire (shielded pair with cam -   but used OEM pin number for clarity.

diy-efi, 2024-03-12:
[1g-fe with cam wired.zip](https://github.com/FOME-Tech/fome-fw/files/14578830/1g-fe.with.cam.wired.zip)

diy-efi, 2024-03-19:
With Dynferus's friend, correcting the VR polarity to match what I have on the PnP he has now had the same issue.
That is 3 confirmed non start / vvt jumping to 120 degree replications on different cars.

A suggestion I made is to add a new setting in advanced trigger to ignore the cam below a certain RPM.

[image]

mck1117, 2024-09-17:
fixed!

closed 2024-09-17 by mck1117