FOME-Tech/fome-fw #383: permissive license or restrictive license?

rusefillc, 2024-03-12:
* https://github.com/FOME-Tech/fome-fw/blob/master/LICENSE is it asking for an s///?
* "f" stands for "free" yet https://github.com/FOME-Tech/fome-fw/blob/master/LICENSE a restrictive license like the General Public License (GPL)?

mck1117, 2024-03-12:
> "f" stands for "free" yet https://github.com/FOME-Tech/fome-fw/blob/master/LICENSE a restrictive license like the General Public License (GPL)?

https://en.wikipedia.org/wiki/GNU_General_Public_License

> The GNU General Public License (GNU GPL or simply GPL) is a series of widely used [free software licenses](https://en.wikipedia.org/wiki/Free_software_license)

closed 2024-03-12 by mck1117

rusefillc, 2024-03-12:
My bad should have created two separate issues https://github.com/FOME-Tech/fome-fw/issues/384