FOME-Tech/fome-fw #384: please remove "rusEFI" from https://github.com/FOME-Tech/fome-fw/blob/master/LICENSE

rusefillc, 2024-03-12:
Please remove most or all mentioning of rusefi from https://github.com/FOME-Tech/fome-fw/blob/master/LICENSE

mck1117, 2024-03-12:
The license specifically states you can't do that.

```
4. Conveying Verbatim Copies.

  You may convey verbatim copies of the Program's source code as you
receive it, in any medium, provided that you conspicuously and
appropriately publish on each copy an appropriate copyright notice;
keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code;
keep intact all notices of the absence of any warranty; and give all
recipients a copy of this License along with the Program.
```

mck1117, 2024-03-12:
By that I mean that _rusEFI's_ license.

rusefillc, 2024-03-12:
Something is _odd_ here. 
* What is the URL for _fome_'s license?
* Should I expect bottom-right corner of the screenshot to point NOT at rusEFI license?

[image]

mck1117, 2024-03-12:
Not sure what the confusion is. US copyright and/or contract law dictates that because FOME is distributing something originally licensed by rusEFI as GPLv3, it must be distributed with the same license. That license contains the word rusEFI. As I am uninterested in learning more about copyright law by force, the word rusEFI is now part of the rusEFI license, and as a result, part of the FOME license.

rusefillc, 2024-03-12:
I truly appreciate that this https://github.com/FOME-Tech/fome-fw/ here goes above and beyond in terms of appropriation for the intellectual property _derived_ from rusEFI.

My confusion is the licensing terms of the _original_ work in this repo which there is many. I am looking to clarify the licensing terms of that original work here, I was assuming to see the top level license as the license for _that_ original content not lua license, rusEFI license or OpenBLT license.

mck1117, 2024-03-12:
Every other fork on GitHub I've ever seen reproduces the license verbatim from its parent.

mck1117, 2024-03-12:
That is to say, distribution of the license itself doesn't make any sense. It's distributed _with_ the rest of the source, which includes a readme that clearly says this is a thing called FOME.