FOME-Tech/fome-fw #388: Unused fueling after start fields

mck1117, 2024-03-14:
Looks like these don't do anything, should they?

```
int8_t[AFTERSTART_HOLD_CURVE_SIZE] afterstartCoolantBins;;"C", 1, 0, -40, 120, 0
uint8_t[AFTERSTART_HOLD_CURVE_SIZE] afterstartHoldTime;;"Seconds", 1, 0, 0, 100, 0
uint16_t[AFTERSTART_ENRICH_CURVE_SIZE] autoscale afterstartEnrich;;"ratio", 0.01, 0, 0, 10, 2
uint8_t[AFTERSTART_DECAY_CURVE_SIZE] afterstartDecayTime;;"Seconds", 1, 0, 0, 100, 0
```

nmschulte, 2024-03-14:
https://github.com/rusefi/rusefi/pull/1230

f08b1542b086eb0de505d55d871f6f6c5448481c: introduces configs, doesn't use them
cbfdd37243016a0e971e9a631b1754b692e793c2: uses `afterstartCoolantBins` in `tsOutputChannels.tuneCrc16`
b27583a2eb7725462d3e3920162d9773155cf3d9: improves `tsOutputChannels.tuneCrc16` to not use `afterstartCoolantBins`

rusefillc, 2024-03-18:
@nmschulte @mck1117 thank you https://github.com/rusefi/rusefi/commit/89c1b68c576de20157b0ec2ba51db3d8194c9895

mck1117, 2024-03-18:
gone! adb5f6bd5c268974256656c32d14942b5df903d8

closed 2024-03-18 by mck1117