FOME-Tech/fome-fw #394: false "FOME Simulator program not found" due to invalid SIMULATOR_BINARY_PATH

rusefillc, 2024-03-20:
Download bundle, unzip bundle
Launch console

Actual result:
console fails to locate simulator
``private static final String SIMULATOR_BINARY_PATH = "../simulator/build";`` seems to be issue?

[image]

nmschulte, 2024-03-20:
Looking, probably due to https://github.com/FOME-Tech/fome-fw/pull/378

nmschulte, 2024-03-20:
I wonder actually if it's the added `canExecute()` check: https://github.com/FOME-Tech/fome-fw/pull/378/files#diff-c4208a0f2492980deaa068a590d5d3fcdbf503ccea7b7e29f8733c7996df39ceR108
```java
        if (!binary.exists() || binary.isDirectory() || !binary.canExecute()) {
```
IIUC, Console will use a different path than that of `SimulatorExecHelper`: https://github.com/FOME-Tech/fome-fw/pull/378/files#diff-bd5c3c893b546d5e8f982a022ab3a552179c982bedf08776b65d42ebde23f190R17
```java
private static final String SIMULATOR_BINARY_PATH = "./";
```

nmschulte, 2024-03-20:
On my machine, when I:
```sh
cd fome.release_2312.f407-discovery/console/
java -jar fome_console.jar
```

I don't see the same behavior:
[image]

nmschulte, 2024-03-20:
Also, not sure why the bundled `fome_console.jar` shows _Version 20230220_ but the `fome_console.exe` shows _Version 20231228_.

mck1117, 2024-03-20:
> On my machine, 

is this a windows machine?

nmschulte, 2024-04-16:
> > On my machine,
> 
> is this a windows machine?

I don't have one of those handy; can someone with a Windows machine give it a whirl?  Otherwise, I'll try to find one this week if I can.

nmschulte, 2024-04-17:
@rusefillc can you see if this is fixed via #395 (snapshot build artifacts here: https://github.com/FOME-Tech/fome-fw/actions/runs/8728754078)?

rusefillc, 2024-04-18:
fixed

[image]

simulator does not launch but that's a different issue

[image]

rusefillc, 2024-04-18:
> 
> simulator does not launch but that's a different issue
> 

simulator.exe has disappeared from folder is it Win11 defender eating it?

closed 2024-04-18 by rusefillc