FOME-Tech/fome-fw #417: m_maximumretard broken, resulting in knock controller not working. Related to Closed Loop Fueling?

rk-iv, 2024-04-30:
Proteus f7 BMM ecu for Mazda Miata NA6

When setting up the Max knock retard table with an array of values ranging from 2 through 10. The variable m_maximumretard does not correlate properly, I assumed if i set up a table of RPM, MAP, Knock: max retard i would see a table that represents my tune table. That was not the case. Furthermore, It made me realize it was not pulling timing when knock count incremented. 

This is what I have found so far

- ECU powered on W/ Closed Loop Fuel Control - ON
              - m_maximumretard is a constant 0 regardless of Max Knock Retard Table
- ECU powered on W/ Closed Loop Fuel Control - ON, then Turned OFF
              -   m_maximumretard changes from 0 to 2 regardless of Max Knock Retard Table
              - Upon power cycle, m_maximumretard value stayed 2
- ECU powered on w/ Closed Loop Fuel Control ON, then OFF, Then ON again
              - m_maximumretard starts at 0, goes to 2.
              - Started the car, m_maximumretard goes to 8, stays at 8 constant regardless of Max Knock Retard Table 
              - Drove around, took a log, both Closed loop fuel control and Knock Controller work as Expected. Pulling timing, and fuel trimming.
              - When I turned the car off, the m_maximumretard stayed constant at 8. When I power cycled ECU, value returned to 0.

nmschulte, 2024-04-30:
see also https://github.com/FOME-Tech/fome-fw/issues/246

rk-iv, 2024-05-01:
M_maximumretard constant zero when ecu is powered on with Closed loop fuel control

[image]

m_maximumretard changes to to constant 2 when CLFC is turned Off.

[image]

Turned CLFC back on, and turned on car, Knock controller is Active

[image]

[image]
[image]

rusefillc, 2024-05-01:
is this https://github.com/rusefi/rusefi/issues/6370?

rk-iv, 2024-05-01:
@rusefillc Let me go give this a try, ill report back

rk-iv, 2024-05-01:
With zero's in the first row, it did not pull timing, and m_maximumretard was Zero when the car was off, and remained zero when the car was on and didnt change regardless of RPM or MAP

It did not make it so that upon ecu power cycle that the m_maximumretard would be anything but constant zero.

changing the bottom cells back to Non zeroes, and doing the step of Closed loop fuel control off then on resulted in the knock controller responding.

mck1117, 2024-05-04:
Not related to closed loop fueling, related to any config burn. If you do a burn, then the knock retard max table would actually work.

nmschulte, 2024-05-24:
@rk-iv is this resolved via #419?

rk-iv, 2024-06-01:
> @rk-iv is this resolved via #419?
yes, latest snapshot has resulted in the max knock table working as expected and not requiring a burn to configure

closed 2024-06-01 by nmschulte