FOME-Tech/fome-fw #425: Idle & Power Region swapped in Closed loop fueling

rk-iv, 2024-05-15:
BMM ecu firmware 20240330

While Idling, changing the Idle Max add and remove percentage in Closed Loop Fueling does not seem to adjust the correct region. However, when Adjusting the Main region, it does works while idling.

My settings are very similar to the above

[image]

[clfc_issue_2024-05-14_07.29.35.zip](https://github.com/FOME-Tech/fome-fw/files/15316598/clfc_issue_2024-05-14_07.29.35.zip)

nmschulte, 2024-05-24:
The log shows that the determined CLF bin is always "main," (`fuelClosedLoopBinIdx` == 3) and presumably this is because the RPM is always > 1000 RPM: e.g. never actually idling.

[image]

mck1117, 2024-08-05:
looks like it's working as intended....

closed 2024-08-05 by mck1117