FOME-Tech/fome-fw #433: cannot calibrate ETB/TPS without accel pedal

nmschulte, 2024-05-25:
`EtbController` won't return successful `init` unless there's an accel pedal present: https://github.com/FOME-Tech/fome-fw/blob/master/firmware/controllers/actuators/electronic_throttle.cpp#L165

This seems unnecessary, for both TPS auto-calibration and PID autotune.

mck1117, 2025-01-01:
That seems like a reasonable restriction, why would somebody run an ETB without a pedal?

nmschulte, 2025-01-02:
[mck1117](https://github.com/mck1117) commented
> why would somebody run an ETB without a pedal?

[nmschulte](https://github.com/nmschulte) commented
> This seems unnecessary, for both TPS auto-calibration and PID autotune.

----

At the time, I had no need for a pedal -- bench testing/development I suppose.

It's no big deal if this restriction remains, but it didn't seem useful to me at the time, even in practice: you can command ETB through TS without a pedal in various ways.

Feel free to close if you don't want to see it change.