FOME-Tech/fome-fw #466: fome-fw doesnt compile on mac with m1 chip

D0kRay, 2024-08-16:
If I run any task from vscode for example 'Build Hellen NB1' it always fails after the bootloader compilation with the following error: **Creating blbuild/fome_bl.list

   text    data     bss     dec     hex filename
  18416      64  131008  149488   247f0 blbuild/fome_bl.elf

Done
../../common_make.sh: line 48: 0xTEXT: value too great for base (error token is "0xTEXT")**

The following toolchain is used: arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi
And this is the commandline output:

 *  Executing task: bash ./compile_hellen-nb1.sh 

Entering common_make.sh with board hellen/hellen-nb1 and CPU ARCH_STM32F4
Board dir is /Users/macuser/Documents/VSCode/fome-fw/firmware/config/boards/hellen/hellen-nb1
FW dir is /Users/macuser/Documents/VSCode/fome-fw/firmware
Calling make for the main firmware...
found /Users/macuser/Documents/VSCode/fome-fw/firmware/config/boards/hellen/hellen-nb1/connectors/generated_ts_name_by_pin.cpp 
PROJECT_BOARD: hellen/hellen-nb1
BOARD_DIR:     /Users/macuser/Documents/VSCode/fome-fw/firmware/config/boards/hellen/hellen-nb1
PROJECT_CPU:   ARCH_STM32F4
SHORT_BOARD_NAME: hellen-nb1
CPU_HWLAYER:   ports/stm32/stm32f4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
UNAME_SM:          Darwin arm64
COMPILER_PLATFORM: arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi
TRGT:              ./ext/build-tools/arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi/bin/arm-none-eabi-
echo "#pragma once" > gitversion.h.gen
printf '#include "fan_control.h"\n #include "fuel_pump.h"\n #include "gear_detector.h"\n #include "trip_odometer.h"\n' > engine_modules_generated.h.gen
printf 'FanControl1,FanControl2, FuelPumpController, GearDetector, TripOdometer,' > modules_list_generated.h.gen
echo "#define GIT_HASH \"$(git rev-parse HEAD)\"" >> gitversion.h.gen
rsync --checksum engine_modules_generated.h.gen engine_modules_generated.h
rsync --checksum modules_list_generated.h.gen modules_list_generated.h
Compiling pch.h
echo "#define GIT_HASH_SHORT \"$(git rev-parse --short HEAD)\"" >> gitversion.h.gen
rsync --checksum gitversion.h.gen gitversion.h
Compiling generated_ts_name_by_pin.cpp
Compiling ch.cpp
Compiling null_device.cpp
Compiling compressed_block_device.cpp
Compiling mass_storage_device.cpp
Compiling mass_storage_init.cpp
Compiling lua.cpp
Compiling lua_hooks.cpp
Compiling can_filter.cpp
Compiling lua_hooks_util.cpp
Compiling script_impl.cpp
Compiling output_lookup.cpp
Compiling value_lookup.cpp
Compiling lua_can_rx.cpp
Compiling state_sequence.cpp
Compiling big_buffer.cpp
Compiling engine_math.cpp
Compiling speed_density.cpp
Compiling closed_loop_fuel.cpp
Compiling closed_loop_fuel_cell.cpp
Compiling lambda_monitor.cpp
Compiling throttle_model.cpp
Compiling ford_aspire.cpp
Compiling custom_engine.cpp
Compiling bmw_m73.cpp
Compiling bmw_m73_mre.cpp
Compiling bmw_n73.cpp
Compiling mazda_miata.cpp
Compiling mazda_miata_base_maps.cpp
Compiling mazda_miata_1_6.cpp
Compiling mazda_miata_na8.cpp
Compiling mazda_miata_vvt.cpp
Compiling mazda_miata_custom_hunchback.cpp
[... 357 lines cut ...]
Compiling util_dummy.cpp
Compiling crc.cpp
Compiling fragments.cpp
Compiling math.cpp
Compiling pt2001.cpp
Compiling openblt_chibios.cpp
Compiling openblt_flash.cpp
Compiling openblt_usb.cpp
Compiling bootloader_stubs.cpp
Compiling bootloader_main.cpp
Linking blbuild/fome_bl.elf
lto-wrapper: warning: Options to '-Xassembler' do not match: -alms=blbuild/lst/crt1.lst, -alms=blbuild/lst/hal.lst, dropping all '-Xassembler' and '-Wa' options.
Creating blbuild/fome_bl.hex
Creating blbuild/fome_bl.bin
Creating blbuild/fome_bl.dmp
Creating blbuild/fome_bl.list

   text    data     bss     dec     hex filename
  18416      64  131008  149488   247f0 blbuild/fome_bl.elf

Done
../../common_make.sh: line 48: 0xTEXT: value too great for base (error token is "0xTEXT")

 *  The terminal process "/bin/zsh '-l', '-c', 'bash ./compile_hellen-nb1.sh'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

D0kRay, 2024-08-16:
the following version of objdump is on my mac:
    objdump --version
    Apple LLVM version 15.0.0
      Optimized build.
      Default target: arm64-apple-darwin23.5.0
      Host CPU: apple-m1
    
      Registered Targets:
        aarch64    - AArch64 (little endian)
        aarch64_32 - AArch64 (little endian ILP32)
        aarch64_be - AArch64 (big endian)
        arm        - ARM
        arm64      - ARM64 (little endian)
        arm64_32   - ARM64 (little endian ILP32)
        armeb      - ARM (big endian)
        thumb      - Thumb
        thumbeb    - Thumb (big endian)
        x86        - 32-bit X86: Pentium-Pro and above
        x86-64     - 64-bit X86: EM64T and AMD64

D0kRay, 2024-08-16:
now it compiles completely.
The following stuff needs to be installed to compile fome one macos sonoma

brew
	brew install mtools
	brew install dosfstools

compile hex2dfu for arm cpu
	gcc hex2dfu.c -o hex2dfu.bin

common_make.sh
	$HEX2DFU -i build/fome.hex -c $checksumAddress -o build/fome.bin

mck1117, 2024-08-16:
I don't think this is completed? We can improve the script to do the right things here. The arm-none-eabi compiler is downloaded in a submodule, so if we do the same trick as we do in the makefile (detect what system you're on and select compiler appropriately), we can use the bundled objdump too.

D0kRay, 2024-08-16:
I make a PR with the necessary Changes :)