FOME-Tech/fome-fw #467: No accel after RPM Hard Limit does not do anything

DenisMitchell, 2024-08-16:
https://github.com/FOME-Tech/fome-fw/blob/81b59f70a82911072db0de8e410bcb167f53184d/firmware/integration/rusefi_config.txt#L1019

Text references TODO #413 but that is not associated with this repository. I see no evidence that any code exists for the strategy that this setting describes, and I only see it referenced within the codebase here:
https://github.com/FOME-Tech/fome-fw/blob/81b59f70a82911072db0de8e410bcb167f53184d/firmware/controllers/algo/engine_configuration.cpp#L469

rusefillc, 2024-08-16:
https://github.com/rusefi/rusefi/issues/413

mck1117, 2024-08-16:
Honestly I'm not sure of the point of that setting, and we should maybe just remove it since it seems everyone has been fine without it.

commit 8633d7ce (2024-08-16): dead config entry fix #467
closed 2024-08-16 by mck1117