FOME-Tech/fome-fw #468: Idle Air Control Percentage changes from open loop setpoint with MAP/TPS input during coasting

stevenuki, 2024-08-18:
While attempting to take logs for VE Analysis I had issues with both MLG VE Analyze and TS VE autotune properly adjusting the VE table during coasting or low-TPS low-MAP operation. Datalog inspection found that the IACV position percentage is changing with the idle: coasting flag, even though the options to use a coasting IACV table (and other options like DFCO) are disabled. It does not appear that the idle is entering closed loop during coasting, which leads me to believe that something else is affecting the IACV position that I can not find settings for. Current tune + log attached. 

What I've noticed is at 0% TPS, the IACV position stays at 32 (which is the base idle percentage for open loop), but with a tiny bit of throttle thrown in (and in most cruising and WOT positions), it ramps up to 38. 

"... out VE tuning today again since I decided to try and get that all smoother. I've been having some lean-rich spiking issues on coasting decel, DFCO turned off, closed loop fuel correction turned off, turned off the idle coasting table option so it should be using the base IACV position on decel. For some reason, however, it's not. "

[image]

[IACV issue Tune + Log.zip](https://github.com/user-attachments/files/16651900/IACV.issue.Tune.%2B.Log.zip)

alrijleh, 2024-08-26:
You have “Extra air if throttle pressed” set to 6.  I suspect that’s the IACV behavior you’re seeing.  Could you set it to 0 and see if the issue is fixed?

stevenuki, 2024-08-26:
> You have “Extra air if throttle pressed” set to 6. I suspect that’s the IACV behavior you’re seeing. Could you set it to 0 and see if the issue is fixed?

I found that setting and changed it to 0. That appears to have made the issue go away. The way the tool tip read made me think that the extra percentage was only added during the transient period between idle and coasting / cruise / all the other cells.

alrijleh, 2024-08-26:
Nice.  I added “Extra air if throttle pressed” to the list I created here: https://github.com/FOME-Tech/fome-fw/issues/472