FOME-Tech/fome-fw #469: Weird rpm spikes on Proteus and latest snapshot

dynfer, 2024-08-19:
Overview:
1UZ-FE
36-2 VR crank into proteus VR1
single tooth cam into proteus VR2

RPM happens to drop below current real value happens at idle around 800-> 200rpm and during driving and steady conditions.
[image]

[długie_2.zip](https://github.com/user-attachments/files/16665206/dlugie_2.zip)

dynfer, 2024-08-19:
[image]
[image]

mck1117, 2024-12-18:
did this get resolved with better hardware?

dynfer, 2024-12-18:
> did this get resolved with better hardware?

Yes, the issue was caused by toyota shared ground. Separate cam sensor harness resolved the issue. Max9924 ic

closed 2024-12-18 by dynfer