FOME-Tech/fome-fw #472: Add or improve tooltips in Tunerstudio

alrijleh, 2024-08-26:
There are a number of menus and settings in Tunerstudio that are frequently the source of questions for new users.  Adding contextual help will make FOME easier to use and prevent the same questions from being asked by users.  Also adding links to the wiki where applicable for more in depth information would be good

Settings that need tooltips:
- MAP estimate table
- Barometric pressure correction
- Injection Phase
- TPS/TPS extra fuel
- TPS?TPS AE RPM COrrection

Settings with tooltips that should be improved:
- Idle settings (derivativeFilterLoss and antiwindupFreq)
- Charge temperature Estimation (explain how this is used in fuel calculation)
- Idle Target RPM

OrchardPerformance, 2024-09-18:
Related to this there has been quite a few tool tips updated for the new release. 

I'll try to add in bits when I have a bit of time but certainly don't let that stop anyone else adding to them.

kemes, 2025-06-26:
CAN: The base address setting could include more information and a tooltip.

Controller > CAN Bus > FOME CAN Data Broadcast:
Base address [default: 512]

Created a pull request for this.