FOME-Tech/fome-fw #478: Vary DFCO MAP threshold based on current RPM

alrijleh, 2024-08-31:
I've noticed with my car (NA Miata) that at 4000 rpm the car will decelerate at around 20kpa, while at 2000 it will be closer to 30.  In my experience, setting the DFCO threshold at 24 will result in the car never entering DFCO at low engine speeds, but 28 will result in a jolt when fuel cuts at high engine speed.

Adding an option to set the DFCO MAP threshold on a curve based on RPM would solve the issue, allowing for fuel saving at low engine speed, and smoother behavior at high engine speeds.

DenisMitchell, 2024-09-03:
Can this be avoided by using the DFCO delay?

alrijleh, 2024-09-03:
I currently have a small amount of DFCO delay (0.3 seconds) in place, and still observe this behavior.  Another different approach to improving DFCO would be adding hysteresis to the MAP threshold like we do with RPM. ie cut below 22 MAP and enable fuel above 25 MAP

closed 2024-09-13 by mck1117