FOME-Tech/fome-fw #484: Logs don't record time in less than second increiments

alrijleh, 2024-09-06:
In past releases, logs would show the time to the thousandths of a second.  In the current snapshot, logs only show second increments, making it more difficult to examine high speed events.

Observed on hellen64_NA6

You can see that the timescale at the bottom of the log is only is x.000 seconds
[image]

mck1117, 2024-09-13:
Everything looks fine on my end using a recent log and the same version of MLV...

mck1117, 2024-09-13:
ah nope the bug is real and it's in 9efd565d296737575c5fe5afe6baf982143978e9

mck1117, 2024-09-13:
@alrijleh Should be fixed - give it a shot on a fresh snapshot and see what you see.

closed 2024-09-13 by mck1117

alrijleh, 2024-09-19:
Verified fixed in the 2409 release