FOME-Tech/fome-fw #504: Disable TPS Error check on Auto Calibration

dynfer, 2024-10-30:
Hello,
Suggestion to disable TPS error check during the auto calibration procedure. This would allow Toyota installs to use it.
Currently on real way is to calibrate the TPS manually.

mck1117, 2024-10-30:
what specific behavior do you see that you think is wrong? screenshot/vid?

dynfer, 2024-10-31:
The throttle just stops at random values due to the TPS error, its not moving smoothly from 0 to 100%.

rusefillc, 2024-11-13:
> The throttle just stops at random values due to the TPS error, its not moving smoothly from 0 to 100%.

vid? logs? what throttle is that?

dynfer, 2024-11-13:
@rusefillc I've mainly encountered it with Toyota throttles which require the special setting for the secondary sensor. It basically errors out if setting isn't set right preventing the calibration from finishing.

dynfer, 2024-12-05:
Solved in #514

dynfer, 2024-12-05:
Solved in #514

closed 2024-12-05 by dynfer