FOME-Tech/fome-fw #543: VE Analyze broken

dynfer, 2025-01-25:
Hello,

After using some logs with 12.12.2024 firmware ve analyze is broken.

The problem stems from the fact we dont have an AFR or Lambda field anymore.
Currently we have Air/Fuel Ratio and Lambda1/2/3/4 which are not correctly picked up by MLV since it requires specific names ie. AFR or Lambda.

Currently a workaround is to link the fields in MLV.

nmschulte, 2025-01-25:
This seems to be caused by the change in `[DataLog]` entries: https://github.com/FOME-Tech/fome-fw/commit/f3749ebc82112f33798ff909fd1a2f260420fb08#diff-89e99629b44f02b9bfdaf728a90cc76679eb4d692810acdfdd04776acf2c7cd4L4502

no `Lambda`-titled entry anymore
it seems `AFR`-titled entry may never have existed/worked "correctly" as suggested by this issue

[image]

dynfer, 2025-01-25:
> This seems to be caused by the change in `[DataLog]` entries: [f3749eb#diff-89e99629b44f02b9bfdaf728a90cc76679eb4d692810acdfdd04776acf2c7cd4L4502](https://github.com/FOME-Tech/fome-fw/commit/f3749ebc82112f33798ff909fd1a2f260420fb08#diff-89e99629b44f02b9bfdaf728a90cc76679eb4d692810acdfdd04776acf2c7cd4L4502)
> 
> no `Lambda`-titled entry anymore it seems `AFR`-titled entry may never have existed/worked "correctly" as suggested by this issue
> 
> [image]

Either AFR or Lambda is a valid channel for VE Analyze.

commit 3089a053 (2025-02-02): fix #543
closed 2025-02-02 by mck1117