FOME-Tech/fome-fw #567: Analog input is dead after pin conflict

dynfer, 2025-03-06:
Fw "rusEFI (FOME) release_250220.2025.03.03.proteus_f4.4291694389"
Board Proteus F4

Clear fw after full erase and base map

Set a analog input like MAP to an input set by something else like Lambda (example map on AV5 and lambda on AV6).
A conflict occurs, remove lambda from AV6 and set MAP to AV6 and perform a reboot.
Now you wont get any readings from AV6 by all other sensors, when You switch MAP to any other input it works but AV6 is unusable by any sensor.

mck1117, 2025-03-07:
Probably fixed in 91689a3, @dynfer have a check?

NoTwixD, 2025-03-07:
fixed by [91689a3](https://github.com/FOME-Tech/fome-fw/commit/91689a3b012b42c4ca7de75264a51e99b2cb3823)

closed 2025-03-07 by mck1117