FOME-Tech/fome-fw #588: Fuel Pump Runs when USB is connected with Febuary Release

classicdaily, 2025-04-24:
Fuel Pump runs when ECU is powered up via USB.  Back dated to 2409 September Release with same hardware and issue does not happen.

mck1117, 2025-06-29:
@classicdaily which firmware, and which pin were your fuel pump and main relay assigned to?

mck1117, 2025-06-30:
reproed using 2502 release on polygonus f7, assigned fuel pump/main relay to IGN1/2 and see this

[image]

mck1117, 2025-06-30:
Looks like it's fixed in mainline.

mck1117, 2025-06-30:
This was fixed by ea3b865602f59bacfbf536772865329f023c4ebb

The default battery input pin was briefly set to an output being driven high during startup, so the very first sample indicated ~6 volts then settled to the correct (zero-ish) reading. That commit fixes the initial pin state so it doesn't do that.