FOME-Tech/fome-fw #601: A/C Relay stays on even if CLT > 'Max CLT'

c0nvexo, 2025-05-26:
_Tested on FOME September 2024 Release_

## Steps to reproduce:
1. Set 'Max CLT' to the desired value under Air Conditioning. 
2. Turn on A/C Switch when CLT is colder than 'Max CLT'
3. Wait untill CLT > Max CLT
4. Notice that the A/C still runing. (not expected)
5. Turn A/C switch off and back on.
6. Notice A/C relay won't kick in. (as expected)

## Expected behavior:
I assume, If A/C Relay was is on (this means A/C conditions were already met before) but suddenly CLT > 'Max CLT'. Then disable A/C Relay output.
If user requests A/C Switch on but CLT > Max CLT. Don't enable A/C Relay output. 

[image]

nmschulte, 2025-09-03:
In this case, did the `engineTooHot` output become set?

The A/C Max CLT setting has a 5 °C "deadband" -- the value must go outside of this range before it will take a new value.  However, that shouldn't be the culprit here unless between the time you took note and switched the A/C switch back on, the value was updated.  Logs (`engineTooHot`, `acButtonState`, `acCompressorState`/`acRelay`) would tell.

Looking at the logic currently, it's not clear why this wouldn't be working.

c0nvexo, 2025-09-05:
I wish i had a log from that day, but yesterday I was logging to do some fueling corrections and coincidentially I had some air in the cooling system which made the temps spike up for a bit. Then I read the logs and everything seems to be working as expected . I'm running Snapshot 20250703 now.

I think what I was noticing back then was that the value wasn't updating due to the 5deg deadband required for it to take the new value. And manually switching the a/c switch on and off triggered the update. By reading yesterday logs and tune i noticed i had my Max CLT set at 99deg and it was not untill ~104 that it triggered the engineTooHot value change. 

[image]

[image]

Should that have a deadband to get into the engineTooHot state or only to get out of it so that it waits for that 5deg cooldown to engage the a/c back again? Also that might be something worth to be put inside the tooltip or documentation for the user to be aware of. 

@nmschulte