FOME-Tech/fome-fw #602: RAM3 is disputed territory on F4

dron0gus, 2025-05-27:
On the one hand RAM3 is defined in linker file as part of ram0
https://github.com/FOME-Tech/fome-fw/blob/68ae531552692285742c6e116da673ea74b67a2c/firmware/hw_layer/ports/stm32/stm32f4/STM32F4.ld#L53
On other hand it is hacky-way given to Lua.
https://github.com/FOME-Tech/fome-fw/blob/68ae531552692285742c6e116da673ea74b67a2c/firmware/controllers/lua/lua.cpp#L380

This cause two different issues:
1. on F407 linker can put something to ram3 that actually does not exits. Not sure if F407 Proteus exist. mk file says all Proteus are 427
https://github.com/FOME-Tech/fome-fw/blob/68ae531552692285742c6e116da673ea74b67a2c/firmware/config/boards/proteus/proteus-base.mk#L5
2. on F427 linker can (and already does) put something to ram3 that later can be overwriten by Lua heap.

dron0gus, 2025-05-27:
Ok, total mess.
May be I was confused by EFI_IS_F42x vs IS_STM32F429 and STM32F429xx.
And if there is no Proteus with F407 you have no issue here.

mck1117, 2025-06-04:
F407 proteus doesn't exist, they're all F42x/F76x